VlasovLabTech/6-AXIS-USBCNC-GRBL #1: Spindle enable Bit

cncPan, 2018-12-30:
Hello
I am sending M3 , M5 commands but i can not see PB1 bit (spindle enable bit) to toggle.

Alfrederson, 2018-12-30:
@yaroslavVl I told you it was going to be a problem

@cncPan  check this: https://github.com/usbcnc/grbl/issues/63

cncPan, 2018-12-30:
Thank you for the comment . I configured $30 = $31 = 1000 , min and max spindle speed the same value and used spindle_pwm_bit as enable bit.

yaroslavVl, 2018-12-30:
Ok, I'll fix it.
Now we have next logic:
- PWM+direction (def VARIABLE_SPINDLE, undef USE_SPINDLE_DIR_AS_ENABLE_PIN) or
- PWM+enable (def VARIABLE_SPINDLE, def USE_SPINDLE_DIR_AS_ENABLE_PIN) or
- enable+direction ((doesn't work now. bug) undef VARIABLE_SPINDLE, undef USE_SPINDLE_DIR_AS_ENABLE_PIN) or
- just enable (undef VARIABLE_SPINDLE, def USE_SPINDLE_DIR_AS_ENABLE_PIN)
Is it ok if we will use PWM+enable+direction instead of PWM+enable?

Alfrederson, 2018-12-31:
@yaroslavVl I guess so, since it doesn't leave either pin non-functional

yaroslavVl, 2019-01-02:
Ok, now PWM, Direction and Enable are avalaible by default. You can disable PWM (VARIABLE_SPINDLE) or Enable (USE_SPINDLE_ENABLE_PIN) or both, but direction will always run.

closed 2019-01-02 by yaroslavVl