VlasovLabTech/6-AXIS-USBCNC-GRBL #2: Using Serial1 and USB at the same time?

Alfrederson, 2019-01-02:
Today I was wondering what would it take to enable both USB and Serial1 at the same time. Enabling Serial1 just for sending a duplicate of the report would be trivial (just uncomment #ifndef USEUSB in line 177 of serial.c and uncomment #ifndef USEUSB in line 104 of main.c, which calls USART1_Configuration(115200);). This would be useful for providing data for something like a DRO or sending it via bluetooth to an android/tablet, but what if I wanted to use **both** at the same time? Would I need to keep separate buffers for USB rx data and for Usart1 so text coming from both don't get mixed up? With this it would be possible to use a cellphone for control without having the machine plugged to a computer, or even make it wifi-ready using ESP3D, while at the same time being able to control from a PC.

yaroslavVl, 2019-01-03:
This is complex and unpredictable way to use it))

MonkeyPigCNC, 2019-02-25:
If the reporting was mirrored to a serial port you could easily use that to update a seperate Arduino to display the position and status. I'm have not looked into this as I'm currently designing a 5 axis board for this but that would be really useful to pretty up the front panel of my 3040, and I could use a cheap LCD shield with touch on an uno to also implement the feed hold/reset etc buttons.

Alfrederson, 2019-02-26:
> If the reporting was mirrored to a serial port you could easily use that to update a seperate Arduino to display the position and status. I'm have not looked into this as I'm currently designing a 5 [...]

Monkey, have you tried compiling this firmware for another mcu in the same line with more GPIO's like RBT6? 5 axes leaves very few pins on the C8T6

MonkeyPigCNC, 2019-02-26:
No, I only have a blue pill stm32 at the moment, but I was not wanting to
send anything back, just use the reporting to show work position on the
display, and have some icons on it to use the touchscreen as the 'buttons'.
Not sure at all if it would work at all but as I only have the schematic
drawn at the moment I could move a couple of pins to free the serial port I
think. I was thinking that having a seperate Arduino or bluepill to display
would not mess with anything grbl is doing actually running the machine,
but just do it passively. I'm a hardware engineer mostly and only really
dabble in software as a hobby. I'm sort of just thinking what I can do with
the bits I have at present, and build from there.

Regards Simon

mstrens, 2019-03-23:
I would also be interested having the possibility of using USB or UART without having to reflash the STM32.
For a group of makers, we made a pcb to support the blue pill, interface it to TB6600 (via ULN2803), add a relay for router switching, and add RC filter on each imputs (limits,...).
It is foreseen that this project could be controlled from the PC (via USB) or from another stand alone ESP32 microcomputer (via UART).
It would be nice to be able to flash only once the firmware into the blue pill for both config.

mstrens, 2019-03-23:
I made changes in order to support USB and UART at the same time.
In fact it is not "at the same time" because if you send characters to GRBL via the 2 ways simultanously, GRBL will mix all received characters and will fail.
So, send the data to GRBL only from one source at the time.
GRBL sent his replies to both canals.

The code is available at (in branch USB_and_SERIAL 
https://github.com/mstrens/grbl_6axis_on_stm32/tree/USB_and_SERIAL

Please take care that:
- I did not tested this code.
- my published version has a modified cpu_map.h file because I use different pins for dir and step signals. 

If this code is OK, it would be quite easy to mofify it in order to let the user select between 3 options USEUSB, USEUART and USEBOTH.

mstrens, 2019-03-31:
Please note that the code in previous message is not OK.
In the mean time, I made change on my site in the master branch in order to use TX buffer and interrupt when sending data. Currently the same TX buffer is used for UART or USB. So it is not possible to use both at the same time. This would require using 2 different Tx buffers.
Furthermore it could be that the program locks if USB Tx buffer becomes full if USB is not up and running