VlasovLabTech/6-AXIS-USBCNC-GRBL #4: Modify Virtual_Com_Port_ConfigDescriptor to allow for more power?

Alfrederson, 2019-01-12:
Spindle enable bit and dir bit works ok, I tested them with a brushless spindle motor driver, now there's something that is bugging me. I'm powering a small relay board with 5V coming from the USB Port itself, when either relay is turned on I get a significant voltage drop on the 5V line. Like, with no relay on I get 4.8v, each relay drops the voltage by about 0.2v. Reading into the code, I found this part.

The STM32 identifies itself as a Virtual Com Port, so there's a Configuration Descriptor for it. The byte that specifies how much power the circuit can draw is set at 0x32, or 50 in decimal. This is in **usb_desc.c** , line **67**

According to this page:
https://www.beyondlogic.org/usbnutshell/usb5.shtml#ConfigurationDescriptors

The amount of current the circuit is allowed to draw is that number times 2 mA, 50 times 2 is just 100 mA.

Now, could this be the reason why just a few puny relays are causing that voltage drop? I'll change that value to 7D in order to pull up to 250mA and see what happens

martin2250, 2019-02-07:
The USB descriptor does not affect how much power the PC supplies to the device. At best it will change the over current limit on that port. This is just caused by ohmic losses in your cable, try using the USB cable that came with your phone, it should drop less voltage when the relay is powered on.
Assuming ~100mA per relay, the combined resistance of the +5V and GND wires would be 2 ohms, which is a reasonable value for USB cables.
The best solution to this would be to power the relays from an external source, eg. add a stepdown converter that provides power to the relays from your stepper power supply.

Alfrederson, 2019-02-17:
> The USB descriptor does not affect how much power the PC supplies to the device. At best it will change the over current limit on that port. This is just caused by ohmic losses in your cable, try [...]

The base board I made has its own USB connector, so I used a standard inkjet printer USB cable. Maybe it has too much resistance on the 5V conductor.

yaroslavVl, 2019-02-17:
I have a 5.07V without addition load,
4.66V with 20 Ohm load
and 4.35V with 10 Ohm load.
So It's about 440mA. I have the same conditions with empty mcu flash and without any mcu or other devices. It seems like both **wire** and host **usb** controller **internal resistances**.
For your purposes use the external power supply.

closed 2019-02-17 by yaroslavVl