VlasovLabTech/6-AXIS-USBCNC-GRBL #8: Rotary Encoder Jogging

martin2250, 2019-02-13:
Hi Yaroslav,

first of all thanks for the work you put into this, it's great to have an STM32 version of grbl that actually receives support!

Something that I've always wanted for my machine was a proper jog wheel. I've bought the hardware (one of [these MPG wheels](http://i.ebayimg.com/00/s/NTAwWDUwMA==/z/lD8AAOxy7nNTVi8q/$_3.JPG?set_id=2)) to tinker around. I tried connecting the wheel to a separate Arduino Uno and sending the position via serial, feeding that into a modified version of OpenCNCPilot and using jog commands to execute that on the machine. This always resulted in very jerky motion, sometimes the machine jumped back and forth etc.

I think a better way to implement a hand wheel would be to connect the MPG to grbl directly.
Here's what I have in mind:
- the MPG wheel is connected to two pins on the STM32 (it's a regular quadrature encoder). A hardware timer is used to decode the signal (see the last piece of code [here](http://embeddedsystemengineering.blogspot.com/2016/07/arm-cortex-m3-stm32f103-tutorial.html))
- a new state "HAND_JOG" is added to grbl, which is enabled/disabled with a realtime command or some M-code by the GUI. The MPG is only active in this mode to prevent the user from interfering with normal machine operation. The GUI also chooses the axis, step size and maximum speed (maybe introduce a new G-Code like "G99 X1 F1000", which chooses the X axis with 1mm steps etc.)
- Behavior:
  - turning the wheel by one increment moves the machine by one step (user selected distance, not machine steps)
  - turning the wheel more than one increment moves the machine by x number of steps, with a feed rate which is selectable. If the distance specified by the jog wheel is some distance Y away from the current machine position, stop accepting new pulses from the wheel so the machine always comes to a stop Y mm after the user stopped turning the wheel. (aka if the user turns the wheel very quickly, have it behave like a constant speed jog instead of a fixed distance jog)
  - when the user turns the wheel in the opposite direction than the machine is currently moving, cancel the jog motion and wait for a few milliseconds before accepting new input

I've already tried my hands at implementing this myself, but I don't have the knowledge of grbl's core to pull it off. Should this actually become a thing, you can bet that OpenCNCPilot will be among the first GUIs to support it :)

Cheers!
Martin

mstrens, 2019-02-17:
FYI I wrote a STM32 program that can be used to control a 3 axes CNC running GRBL.
Here the link
https://github.com/mstrens/grbl_controller_stm32
Look at the readme for more details.
It allows to use a Nunchuk ( kind of joystick for a game computer) to move the 3 axis.
It starts slowly and accelerates up to fast movement if the joystick remains in the same poisition.
It also allows the move using a rotary switch.

I also wrote a similar program running on ESP32 but this one uses a touch screen instead of the rotary switch (and it allows to upload files on the Sd card with the wifi.
This version is here (but it is only a beta version; test are ongoing).
https://github.com/mstrens/grbl_controller_esp32
.

yaroslavVl, 2019-02-17:
>  I don't have the knowledge of grbl's core to pull it off

Me too )))