VlasovLabTech/6-AXIS-USBCNC-GRBL #12: Issue with USB

mstrens, 2019-03-25:
When I configure GRBL with USEUSB (so using USB and not UART1), it seems that USB connection locks if I send a Soft Reset or if I push on the Reset button on the blue pill.

I faced this issue following e.g. this process:
- I connect blue pill to pc via USB
- I open a serial terminal session (e.g. using arduino Terminal).
- I can sent commands to GRBL (e.g. "?" to know the status and position)
- If GRBL reach a hard limit (those are activated in my configuration), GRBL sent ALARM:1
- To unlock GRBL you normally have to send a Soft Reset (CTRL+X or 0X18). If I try this, GRBL does not react.
- even a hard reset pressing the reset button on the blue pill does not unlock GRBL communication. 
- To continue, it seems that I have to close my terminal session (even close the program) and reopen it.

When I work with a UART connection between PC and blue pill, GRBL reacts (as expected) after a soft reset.

Do you experience the same issue?

robomechs, 2019-03-26:
Hmm((
It's look like the same issue:
http://www.chibios.com/forum/viewtopic.php?t=1727
Unfortunately

robomechs, 2019-03-26:
[image]
[image]

robomechs, 2019-03-26:
I'll try it later

Alfrederson, 2019-06-04:
You need to close the serial connection before restarting the bluepill. It's not like an arduino where hitting the reset button resets just the MCU and the usb-serial chip stays connected.

troth530, 2020-02-04:
I have the same issue and possibly more issues. I have removed 10k resistor and installed a 1.5k for the usb, but I cannot soft reset after an alarm (limit switch is triggered) and reset button does not bring it back up.  Same issue seen on multiple sources of blue pill and black pill (robotdyn).  Maybe a dumb question @Alfrederson , but how do I close the serial connection, then reconnect?  Also I see this issue also on @mstrens branch.  I think it's a blue pill/grbl issue.

Many thanks!

misan, 2020-02-04:
@troth530 Does Ctrl+X keystroke not reset GRBL?

troth530, 2020-02-04:
@misan no, I cannot get Ctrl+X to work on blue pill or back pill (STM32f103C8).  Reset button does not work either.  I basically have to unplug from USB, then plug in again and recently need to reboot the entire PC.  Can sometimes get $X to work to clear an alarm, but soft reset has never worked.  Not sure if I've damaged my hardware or what.

Alfrederson, 2020-02-05:
> I have the same issue and possibly more issues. I have removed 10k resistor and installed a 1.5k for the usb, but I cannot soft reset after an alarm (limit switch is triggered) and reset button does [...]

Just pull the USB cord and plug it back in. Pushing the reset button on the bluepill itself does the same, but you need to close your host before.

troth530, 2020-02-05:
@Alfrederson. Thank you.  I will try closing the software accessing grbl on the PC, then the reset button on Blue Pill/Black Pill and see if it comes back up.  Sounds like this is a STM32 issue that cannot be remedied in Firmware.  

I should probably also mention that I really appreciate all the hard work that goes into the development and if no one thanks you for all of your hard work, then shame on them!

troth530, 2020-02-06:
I verified that if you close the software host you can then disconnect the USB and reconnect to do a hard reset, but the reset button does nothing to clear the alarm and bring it back out of the alarm state.  

Hard reset seems like it's overkill for a basic alarm like triggering a limit switch.  Should I open another issue because this does not seem USB related.

Alfrederson, 2020-02-07:
> I verified that if you close the software host you can then disconnect the USB and reconnect to do a hard reset, but the reset button does nothing to clear the alarm and bring it back out of the [...]

If it's just an alarm, can't you just unlock it through your GRBL host? Mine starts in alarm mode and I unlock it, no need to reset or anything. I think it's a compilation option to have GRBL start in alarm mode.

troth530, 2020-02-10:
@Alfrederson, how exactly do you unlock it.  I cannot get Ctrl-x (ctrl key + x) to work and I cannot get $X to work either.

robomechs, 2020-05-25:
I suppose it can be fixed only by hardware.

troth530, 2020-05-28:
@Alfrederson  if it starts in alarm state homing can be done to get it out of alarm state, I think it's Alarm 9 in that case.  If I recall it does not require Ctrl-X, just needs to go through the homing cycle.  The issue is on hitting a limit switch after homing it gets in an alarm state Alarm 1 and latches up.

@robomechs, thanks for looking into it.  Disconnect from PC and pulling control_reset_bit low does ensure it can come out of alarm state.  Using the reset button does not work.

I still believe something can be done if you can try to repeat this setup you'll have a better chance to debug.