VlasovLabTech/6-AXIS-USBCNC-GRBL #16: big issue in communication with GRBL at least with UART

mstrens, 2019-03-30:
I have a GRBL controller running on a ESP32. It sent commands to GRBL (e.g. based on a SD file, waiting OK between each command) and it send also "?" every 300 msec (to get the current status).

If I let it communicate with the original AVR 328P GRBL version1.1, it works perfectly.
When I try to let it communicate with STM32 using UART, it does not work.
GRBL gives errors saying that there are syntax errors. So, it seems GRBL is loosing some characrers being sent.
I put a digital scope to analyse the exchanged data and timing.
It confirms that my micro is sending the right data but GRBL replies on regular basis with an error 8:
While the original AVR GRBL is always giving a fast answer to a "?" , the STM32 takes from time to time much more time to reply.
I also noticed that when I sent a command, GRBL takes more and more time to give the OK.
Sometime I get the "O" (from OK) and I get the "K" abour 100msec later.
So it seems something is "freezing" GRBL and it loose some char being sent in the UART.

I do not know if someone uses UART and already noticed such an issue.
This is blocking my project currently.

I can reproduce this issue when I send a file like this (whis is the same as the command generated by my micro controller when I use a joystick to move the motors):
`
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y0.01 F2000
$j=G91 G21 Y0.01 F2000
$j=G91 G21 Y0.01 F2000
$j=G91 G21 Y0.01 F2000
$j=G91 G21 Y0.01 F2000
$j=G91 G21 Y0.01 F2000
$j=G91 G21 Y0.01 F2000
$j=G91 G21 Y0.01 F2000
$j=G91 G21 Y0.01 F2000
$j=G91 G21 Y0.01 F2000
$j=G91 G21 Y0.01 F2000
$j=G91 G21 Y0.10 F2000
$j=G91 G21 Y0.10 F2000
$j=G91 G21 Y0.10 F2000
$j=G91 G21 Y0.10 F2000
$j=G91 G21 Y0.10 F2000
$j=G91 G21 Y0.10 F2000
$j=G91 G21 Y1.00 F2000
$j=G91 G21 Y1.00 F2000
$j=G91 G21 Y1.00 F2000
$j=G91 G21 Y1.00 F2000
$j=G91 G21 Y1.00 F2000
$j=G91 G21 Y1.00 F2000
$j=G91 G21 Y1.00 F2000
$j=G91 G21 Y1.00 F2000
$j=G91 G21 Y1.00 F2000
$j=G91 G21 Y1.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000
$j=G91 G21 Y3.00 F2000

`

robomechs, 2019-03-30:
And what's about usb cdc?

mstrens, 2019-03-30:
I have not yet test with USB.
Anyway, with USB I can't trace the message with my digital scope (it is just a cheep chinese item).
Furthermore, to communicate between my ESP32 stand alone micro computer and GRBL I can't use USB.  UART is by far the easiest way to let ESP32 communicates with STM32

robomechs, 2019-03-30:
Ok.

misan, 2019-03-30:
@mstrens There is a native port of GRBL for ESP32 (not for 6 axes though) https://github.com/bdring/Grbl_Esp32

mstrens, 2019-03-30:
I know the ESP32 project but I wanted a system that can run is stand alone with a display, a touch screen and a sd card. ESP32 does not have enough pins to manage all this. My ESP32 project seems OK (just some more tests to finalise) but the issues are now with GRBL on STM32. With some other people we already made some PCB for the Blue pill and for the ESP32. 
I hope we will find the solution for this issue with GRBL on STM32

mstrens, 2019-03-31:
I made some more analysis.
I changed the code in order to set a pin HIGH each time the program enter the interrupt that generates step and dir signals. The pin is then set LOW when it exits the interrupt.
When I run the test Gcode, I saw that I got pulse about 5 usec HIGH and 5 usec LOW.
It means that the signal was at about 100khz.
I expect that this does not leave lot of time to the CPU to manage other tasks (there is less than 5 usec between interrupt).
I looked further the set up and I was using the default values (5200 steps/mm).
This high value combined with a high speed (2000 mm/min), explains that GRBL had to generate signals at such a high frequency and so lose some incomming characters.
I reduced the step/mm to 120 (realistic for my machine) and now it seems OK.
I will further investigate to be sure it is really OK.