VlasovLabTech/6-AXIS-USBCNC-GRBL #21: Feed Overrides doesn't work properly

S2000Stefan, 2019-04-20:
Hello,
Feed Overrides doesn't work properly. Plus, minus 10 percent work well change to 20 percent 
freeze STM32 GRBL. 
 Version "stm32grbl11_11122018" work well, later versions doesn't work.
Greetings Stefan

creisey, 2020-01-12:
i have the same. but the version you mention is no longer longer accessable, the latest i see is form 28.Dec 2018
[image]

S2000Stefan, 2020-01-12:
Hello creisey,
i am work on my 4 achses milling machine with this version from here with small changes:
[https://github.com/usbcnc/grbl/issues/51](url)
It works ok for me.
Greetings Stefan

creisey, 2020-01-12:
thx a lot, ill try yours too