VlasovLabTech/6-AXIS-USBCNC-GRBL #25: invert feed-hold

creisey, 2019-12-09:
hi togheter. 
i have tried so far without any success to invert the signal in the file cpu_map.h

Line 123

  #define CONTROL_MASK      ((1<<CONTROL_RESET_BIT)|(1<<CONTROL_FEED_HOLD_BIT)|(1<<CONTROL_CYCLE_START_BIT)|(1<<CONTROL_SAFETY_DOOR_BIT))
  #define CONTROL_INVERT_MASK   CONTROL_MASK // May be re-defined to only invert certain control pins.

without any success...

how do i do that exactly?

mstrens, 2020-01-14:
I think that there is a version on github (superGerbil) where changes about this are already made
https://github.com/paulusjacobus/SuperGerbil

See change on 29 march 2019

robomechs, 2020-07-29:
you have to do it in config.h line 260, not in cpu_map.h (this is undefined part of code related with AVR)

closed 2020-07-29 by robomechs