VlasovLabTech/6-AXIS-USBCNC-GRBL #33: Issue w/ parser mode state reporting

Duffmann, 2020-05-25:
CNCjs is not able to parse modal state reports by 6-AXIS-USBCNC-GRBL due to a small glitch in [report.c](https://github.com/robomechs/6-AXIS-USBCNC-GRBL/blob/master/src/report.c). Cause is an additional ` report_util_gcode_modes_M();` in Line 318 of report.c here which is not in the original grbl v1.1f
```
  report_util_gcode_modes_M();   <<< Offending statement, will inject a singular ' M' into the report that cannot be parsed
  #ifdef ENABLE_M7
    if (gc_state.modal.coolant) { // Note: Multiple coolant states may be active at the same time.
      if (gc_state.modal.coolant & PL_COND_FLAG_COOLANT_MIST) { report_util_gcode_modes_M(); serial_write('7'); }
      if (gc_state.modal.coolant & PL_COND_FLAG_COOLANT_FLOOD) { report_util_gcode_modes_M(); serial_write('8'); }
    } else { report_util_gcode_modes_M(); serial_write('9'); }
  #else
    report_util_gcode_modes_M();
    if (gc_state.modal.coolant) { serial_write('8'); }
    else { serial_write('9'); }
  #endif
```
It creates this (wrong syntax)
[GC:G0 G54 G17 G21 G90 G94 M0 **M** M9 T0 S0 F0]

instead of this (correct syntax)
[GC:G0 G54 G17 G21 G90 G94 M0 M9 T0 S0 F0]

This Bug will cause [GrblLineParserResultParserState.js](https://github.com/cncjs/cncjs) to ditch all modal reports and not update its grbl info panels accordingly:

```
class GrblLineParserResultParserState {
    // * Grbl v0.9
    //   [G38.2 G54 G17 G21 G91 G94 M0 M5 M9 T0 F20. S0.]
    // * Grbl v1.1
    //   [GC:G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 S0.0 F500.0]
    static parse(line) {
        const r = line.match(/^\[(?:GC:)?((?:[a-zA-Z][0-9]+(?:\.[0-9]*)?\s*)+)\]$/);
        if (!r) {
            return null;
        }
...
```

m0n5t3r, 2020-05-25:
that's present in the usbcnc code as well, and I verified that removing the line fixes the issue

Duffmann, 2020-05-25:
Yes, found the same bug in the forked usbcnc's repo. Anyway, 6-AXIS-USBCNC_GRBL is an excellent piece of work and runs my brand new CNC extremely well. A five grands machine with a $1.99 brain ;-)

robomechs, 2020-05-25:
I'll try to fix it.
Any solution?

m0n5t3r, 2020-05-25:
just remove / comment the line, it fixes the issue

robomechs, 2020-05-25:
done

closed 2020-05-25 by robomechs