VlasovLabTech/6-AXIS-USBCNC-GRBL #34: Makefile?

m0n5t3r, 2020-05-25:
Would it be hard to add a makefile / platformio configuration / whatever else doesn't require a giant IDE (which is also proprietary as far as I can tell)? that's pretty overkill for someone who just wants to compile the firmware once.

The original compiles just fine with the makefile, whu drop it?

m0n5t3r, 2020-05-25:
looks like it should be easy enough according to https://stackoverflow.com/questions/41622318/exporting-atollic-project-as-makefile-based-project

robomechs, 2020-05-25:
I think we can move it to stm32cube ide.
Also project contains .hex file with default 4 axis.

m0n5t3r, 2020-05-25:
Doesn't really help if it's another IDE, the point was to avoid downloading eclipse or bastardizations thereof :)

Anyhow, feel free to close this if it's inconvenient, I've already applied the fixes to the usbcnc code (I already committed its pinout to solder and don't need more than 3 axes - going to keep an eye on this repo, though)

Donny2333, 2025-06-23:
need makefile +1