VlasovLabTech/6-AXIS-USBCNC-GRBL #35: ! news. Project no longer supported.

robomechs, 2020-05-25:
Unfortunatly I'm absolutely have no time to support this project.
In addition, it seems that it is a little outdated.

terjeio, 2020-06-07:
[grblHAL](https://github.com/terjeio/grblHAL) with its [STM103 driver](https://github.com/terjeio/grblHAL/tree/master/drivers/STM32F1xx) could be an alternative. Additional G- and M-codes supported, for STM32CubeIDE.

Duffmann, 2020-06-16:
Thx for all your work on this @robomechs!! Will have a look at grblHAL now.

@terjeio: is the F103 driver of grblHAL pin-to-pin compatible with 6-AXIS-USBCNC-GRBL? 

Final tip from my side: Should someone run into lock-up of USBCNC-GRBL when executing fast, small movements while stepping up/ down Feedrate Override at the same time then try to reduce gcc's optimisation level when building the project:
-o3 works for me, -ofast does lead to GRBL frequent hangs and almost-crashes of my CNC mill

terjeio, 2020-06-16:
> is the F103 driver of grblHAL pin-to-pin compatible with 6-AXIS-USBCNC-GRBL?

not 100%, a new board map may be added file if so required. The [default board](https://github.com/terjeio/grblHAL/blob/master/drivers/STM32F1xx/Inc/driver.h) mapping does not have definitions for the C-axis and I see now that the probe pin is in conflict with A-axis direction pin...